azonenberg/openfpga #47: Bitwidth mismatches

MattEttus, 2016-10-20:
Assignments with mismatched widths.  

Assignments of numbers without specified width (defaults to 32 bits) when the destination is wider than 32 bits.  Been bit by this...

Cool idea, BTW.  Good luck with the project.

cliffordwolf, 2016-10-20:
> (defaults to 32 bits) [..] The 12345 is assumed to be 32 bits wide.

This is not correct. The Verilog standard says that unsized constants must be *at least* 32 bits wide. Yosys creates a larger constant if the number does not fit into 32 bits. After the constant is created it is extended to expression width, which is in this case 64 bits, because in Verilog the LHS size of an assignment contributes to the expression size. So there really ios nothing to warn about here.

The only case that is problematic is in self-sized expressions. But by definition, they don't have a destination size to compare with. (E.g. `{123456789*123456789}` will perform the multiplication with 32 bits width, because all values are 32 bits wide and the expression inside `{...}` is self-sized.)

MattEttus, 2016-10-20:
Yosys may do the smart thing, but the standard is obviously ambiguous.
Other compilers don't necessarily do the smart thing and relying on it is
asking for unportability and trouble.

cliffordwolf, 2016-10-20:
> Yosys may do the smart thing, but the standard is obviously ambiguous.

Yes, the standard is ambiguous, but only for constants that do not fit in 32 bits! There is nothing ambiguous about the code you posted.

cliffordwolf, 2016-10-21:
To further elaborate, I thing the following two conditions should trigger a warning in a Verilog linter:

1. Use of unsized constants in certain self-sized contexts, such as `{...}` or `{N{...}}`, or the argument of a reduce operation.

2. Use of unsized constants with a value that does not fit in 32 bits. (Or maybe even something smaller, like 24 bits, to make sure that there is no overflow that might change the outcome of a calculation, e.g. when adding a few of those constants.)

Verilog rules for expression sizes [are complicated](http://www.clifford.at/yosys/vloghammer_bugs/issue_044_quartus.html)  and [hard to get right](http://www.clifford.at/yosys/vloghammer_bugs/issue_055_verific.html).

closed 2022-10-16 by MattEttus