azonenberg/openfpga #78: Consider vendoring vpi_user.h

rqou, 2017-03-01:
I don't want to have to install Icarus Verilog just to get this one header file.

whitequark, 2017-03-01:
Well gpcosim is not useful without iverilog, is it?

rqou, 2017-03-01:
Theoretically you should be able to use another Verilog simulator along with gpcosim (e.g. a proprietary one). Not sure why exactly you would do that, but...

whitequark, 2017-03-01:
vpi_user.h's license (GPLv2+) is not compatible with openfpga's (LGPLv2.1).

rqou, 2017-08-06:
@azonenberg and I discussed this IRL at DefCon and he mentioned that he isn't a huge fan of VPI and might consider a completely different interface for the cosimulation code. He mentioned that he has/will have something that just uses `$fread` and `$fwrite` on pipes as an example of one possible way to build a new cosimulation framework. The IRL discussion then spiraled into a discussion about whether the potential new framework would be used for cosimulation, mixed-language simulation, or both, and then the discussion turned into a bikeshed about what exactly needs to be expressible in a potential new framework. No conclusions were reached, but I'm just writing this down so that we can remember that this happened.