azonenberg/openfpga #94: segfault if timing.json is missing

rqou, 2017-06-07:


azonenberg, 2017-06-08:
What platform/architecture? Can you provide a core dump or gdb backtrace etc?

Does not crash for me on Debian 8 amd64

rqou, 2017-06-08:
This is really strange. Crashed for me on both x86_64 and PPC Debian Sid, and seems to be an infinite loop in FindCombinatorialPaths:

```
$ gdb ./bin/gp4par
GNU gdb (Debian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./bin/gp4par...done.
(gdb) r --output test.txt ../blinky-46620.json
Starting program: /home/rqou/code/openfpga/build/bin/gp4par --output test.txt ../blinky-46620.json
GreenPAK 4 place-and-route by Andrew D. Zonenberg.

License: LGPL v2.1+
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Loading Yosys JSON file "../blinky-46620.json".
Netlist creator: Yosys 0.7+195 (git sha1 129984e, clang 3.8.0-2ubuntu4 -fPIC -Os)

Loading modules...

Device configuration:
    Target device:   SLG46620V
    VCC range:       not yet implemented
    Unused pins:     float
    User ID code:    00
    Read protection: disabled
    I/O precharge:   disabled
    Charge pump:     auto
    LDO:             enabled
    Boot retry:      1 times

Loading timing data file "../../../timing.json"
Warning: Timing data file not found, unable to do timing-driven placement or evaluate post-PAR timing

Implementing top-level module "Blinky".

Creating netlist graphs...

Optimizing placement...

Building post-route netlist...

Checking post-route design rules...

Device utilization:
    COUNT:         4/10 (40 %)
      COUNT8:      2/ 5 (40 %)
      COUNT14:     2/ 3 (66 %)
    FF:            8/12 (66 %)
      DFF:         6/ 6 (100 %)
      DFFSR:       2/ 6 (33 %)
    IOB:          11/18 (61 %)
    LFOSC:         1/ 1 (100 %)
    LUT:          10/26 (38 %)
      LUT2         4/ 8 (50 %)
      LUT3         5/16 (31 %)
      LUT4         1/ 2 (50 %)
    POR:           1/ 1 (100 %)
[... 1428 lines cut ...]
    basepath=std::vector of length 47156, capacity 47156 = {...}, tip=0x555555c0b870, paths=std::set with 0 elements)
    at /home/rqou/code/openfpga/src/gp4par/par_reporting.cpp:501
#473 0x000055555564402c in FindCombinatorialPaths (sinks=std::set with 8 elements = {...}, 
    basepath=std::vector of length 47155, capacity 47155 = {...}, tip=0x555555c0b9f0, paths=std::set with 0 elements)
    at /home/rqou/code/openfpga/src/gp4par/par_reporting.cpp:501
#474 0x000055555564402c in FindCombinatorialPaths (sinks=std::set with 8 elements = {...}, 
    basepath=std::vector of length 47154, capacity 47154 = {...}, tip=0x555555c0b870, paths=std::set with 0 elements)
    at /home/rqou/code/openfpga/src/gp4par/par_reporting.cpp:501
#475 0x000055555564402c in FindCombinatorialPaths (sinks=std::set with 8 elements = {...}, 
    basepath=std::vector of length 47153, capacity 47153 = {...}, tip=0x555555c0b9f0, paths=std::set with 0 elements)
    at /home/rqou/code/openfpga/src/gp4par/par_reporting.cpp:501
#476 0x000055555564402c in FindCombinatorialPaths (sinks=std::set with 8 elements = {...}, 
    basepath=std::vector of length 47152, capacity 47152 = {...}, tip=0x555555c0b870, paths=std::set with 0 elements)
    at /home/rqou/code/openfpga/src/gp4par/par_reporting.cpp:501
#477 0x000055555564402c in FindCombinatorialPaths (sinks=std::set with 8 elements = {...}, 
    basepath=std::vector of length 47151, capacity 47151 = {...}, tip=0x555555c0b9f0, paths=std::set with 0 elements)
    at /home/rqou/code/openfpga/src/gp4par/par_reporting.cpp:501
#478 0x000055555564402c in FindCombinatorialPaths (sinks=std::set with 8 elements = {...}, 
    basepath=std::vector of length 47150, capacity 47150 = {...}, tip=0x555555c0b870, paths=std::set with 0 elements)
    at /home/rqou/code/openfpga/src/gp4par/par_reporting.cpp:501
#479 0x000055555564402c in FindCombinatorialPaths (sinks=std::set with 8 elements = {...}, 
    basepath=std::vector of length 47149, capacity 47149 = {...}, tip=0x555555c0b9f0, paths=std::set with 0 elements)
    at /home/rqou/code/openfpga/src/gp4par/par_reporting.cpp:501
<many more lines>
```

azonenberg, 2017-06-09:
Seems specific to the Blinky test and happens even if you have the timing data file. Investigating...

azonenberg, 2017-06-09:
Fixed in 48edca9

closed 2017-06-09 by azonenberg