azonenberg/openfpga #95: Regression from hidapi switchover: Multiple dev boards don't seem to work right

azonenberg, 2017-06-08:
azonenberg@skipjack:/nfs4/home/azonenberg/code/openfpga/build$ ./bin/gp4prog --device 1 --blink
GreenPAK 4 programmer by Andrew D. Zonenberg and whitequark.

License: LGPL v2.1+
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Searching for developer board at index 1
    ERROR: No device found, giving up

OLD VERSION (before hidapi switchover)
azonenberg@skipjack:/nfs4/home/azonenberg/code/openfpga/build$ /usr/local/bin/gp4prog --device 1 --blink
GreenPAK 4 programmer by Andrew D. Zonenberg and whitequark.

License: LGPL v2.1+
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Searching for developer board at index 1
    Found: Silego GreenPAK Universal Dev. Board 1
Detecting part
    Detected empty SLG46620V
No actions requested, exiting (use --help for help)

rqou, 2017-06-08:
Can you please run before/after with --debug?

rqou, 2017-06-08:
Er, really strange question: while testing this, did you ever run the old libusb-based version? If so, then this version will not work until you unplug and replug the board. This is because the old code detached the Linux HID driver to get it out of the way, but it doesn't ever attach it back. Once that happens, this version can't access it anymore.

azonenberg, 2017-06-08:
That would explain a lot. Confirmed, that's the problem.

Closing as invalid.

closed 2017-06-08 by azonenberg

rqou, 2017-06-08:
Just for completeness, I tested interactions with the blob. The blob doesn't seem to have problems grabbing the device even if libusb detached the kernel driver, but it also does correctly reattach the kernel driver, so there are no new problems with that.