azonenberg/openfpga #127: xc2par fails on recent Yosys json output

miek, 2019-11-29:
I'm trying out xc2par and following the quick start instructions in @rqou's blog post: https://robertou.com/unofficial-open-source-place-and-route-for-xilinx-coolrunner-ii-cplds.html

I'm running into the following error:

    $ ./target/release/xc2par -p xc2c32a-4-vq44 blinky.json -v
    [...]
    Nov 29 20:05:46.418 INFO cells - processing, type: IOBUFE, cell name: $auto$iopadmap.cc:385:execute$654, top module: top, pass: yosys -> intermediate
    Nov 29 20:05:46.418 INFO cells - processing, type: IOBUFE, cell name: $auto$iopadmap.cc:385:execute$656, top module: top, pass: yosys -> intermediate
    Nov 29 20:05:46.418 INFO cells - processing, type: FDCP, cell name: $auto$simplemap.cc:420:simplemap_dff$56, top module: top, pass: yosys -> intermediate
    Nov 29 20:05:46.418 ERRO cells - parameter not a number, value: , name: INIT, cell name: $auto$simplemap.cc:420:simplemap_dff$56, top module: top, pass: yosys -> intermediate
    Error: FrontendError(IllegalAttributeValue(S("")))

and the relevant part of the json looks like this:

        "$auto$simplemap.cc:420:simplemap_dff$56": {
          "hide_name": 1,
          "type": "FDCP",
          "parameters": {
            "INIT": ""
          },
          "attributes": {
            "src": "blinky.v:22"
          },
          "port_directions": {
            "C": "input",
            "CLR": "input",
            "D": "input",
            "PRE": "input",
            "Q": "output"
          },
          "connections": {
            "C": [ 80 ],
            "CLR": [ "0" ],
            "D": [ 74 ],
            "PRE": [ "0" ],
            "Q": [ 7 ]
          }
        },

I've done a bisect on Yosys and found that YosysHQ/yosys@15fae357f6c7a5c205d2733cd0b2640eecd17a2d is the first commit that causes problems. I don't know enough about the internals to figure out the correct fix, so I'm hoping someone can point me in the right direction. Cheers!

cr1901, 2020-01-29:
@miek While I'm stopping by for my irregular "make sure openfpga still works" test, it looks like the problem line is [here](https://github.com/azonenberg/openfpga/blob/f952d282897dc169b0e139bd9b36574241edbe30/src/xc2par/src/frontend.rs#L692):

```
init_state: numeric_param("INIT")? != 0
```

`INIT` has become a string parameter as of your linked commit in order to handle `z` and `x`. But in `xc2par`, it's only useful for extracting numerical values. I'm not sure the best way to fix this offhand, but I would _personally_ add code to the [linked](https://github.com/azonenberg/openfpga/blob/f952d282897dc169b0e139bd9b36574241edbe30/src/xc2par/src/frontend.rs#L390-L395) `else` clause to do a [`TryFrom`](https://doc.rust-lang.org/std/convert/trait.TryFrom.html) conversion from a string to int.

rqou, 2020-02-09:
Thank you for using and reporting bugs in openfpga! I believe I have fixed this issue in commit 0fe16977b5307840cd5498c2e93f197e93b12f56.

closed 2020-02-09 by rqou