azonenberg/openfpga #130: xc2par: Panic when building for xc2c64a-7-vq100

miek, 2020-02-17:
When trying to build a simple blinky example for the XC2C64A, I get a panic in engine::do_par:

```
$ cat blinky.v 
module top(led0, clk_);

(* LOC = "FB1_1" *)
output led0;
(* LOC = "FB2_10" *)
input clk_;

wire clk;
BUFG bufg0 (
    .I(clk_),
    .O(clk),
);

reg [8:0] counter;

assign {led0} = counter[8];

always @(posedge clk)
    counter <= counter + 1;
endmodule

$ yosys -p "synth_coolrunner2 -json blinky.json" blinky.v
[...]

$ RUST_BACKTRACE=1 ./target/debug/xc2par -p xc2c64-7-vq100 blinky.json -v
Feb 17 03:36:38.047 INFO cells - processing, type: ANDTERM, cell name: $auto$coolrunner2_sop.cc:143:execute$235, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.048 INFO cells - processing, type: ANDTERM, cell name: $auto$coolrunner2_sop.cc:143:execute$238, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.048 INFO cells - processing, type: ANDTERM, cell name: $auto$coolrunner2_sop.cc:143:execute$241, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.048 INFO cells - processing, type: ANDTERM, cell name: $auto$coolrunner2_sop.cc:143:execute$244, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.049 INFO cells - processing, type: ANDTERM, cell name: $auto$coolrunner2_sop.cc:143:execute$247, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.049 INFO cells - processing, type: ANDTERM, cell name: $auto$coolrunner2_sop.cc:143:execute$250, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.049 INFO cells - processing, type: ANDTERM, cell name: $auto$coolrunner2_sop.cc:143:execute$253, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.049 INFO cells - processing, type: ANDTERM, cell name: $auto$coolrunner2_sop.cc:143:execute$256, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.050 INFO cells - processing, type: MACROCELL_XOR, cell name: $auto$coolrunner2_sop.cc:154:execute$236, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.050 INFO cells - processing, type: MACROCELL_XOR, cell name: $auto$coolrunner2_sop.cc:154:execute$239, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.050 INFO cells - processing, type: MACROCELL_XOR, cell name: $auto$coolrunner2_sop.cc:154:execute$242, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.050 INFO cells - processing, type: MACROCELL_XOR, cell name: $auto$coolrunner2_sop.cc:154:execute$245, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.051 INFO cells - processing, type: MACROCELL_XOR, cell name: $auto$coolrunner2_sop.cc:154:execute$248, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.051 INFO cells - processing, type: MACROCELL_XOR, cell name: $auto$coolrunner2_sop.cc:154:execute$251, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.051 INFO cells - processing, type: MACROCELL_XOR, cell name: $auto$coolrunner2_sop.cc:154:execute$254, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.051 INFO cells - processing, type: MACROCELL_XOR, cell name: $auto$coolrunner2_sop.cc:154:execute$257, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.052 INFO cells - processing, type: ANDTERM, cell name: $auto$coolrunner2_sop.cc:285:execute$260, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.052 INFO cells - processing, type: MACROCELL_XOR, cell name: $auto$coolrunner2_sop.cc:292:execute$261, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.052 INFO cells - processing, type: IBUF, cell name: $auto$iopadmap.cc:403:execute$263, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.052 INFO cells - processing, type: IOBUFE, cell name: $auto$iopadmap.cc:403:execute$265, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.052 INFO cells - processing, type: FDCP, cell name: $auto$simplemap.cc:420:simplemap_dff$51, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.053 INFO cells - processing, type: FTCP, cell name: $extract$\FTCP$217, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.053 INFO cells - processing, type: FTCP, cell name: $extract$\FTCP$218, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.053 INFO cells - processing, type: FTCP, cell name: $extract$\FTCP$219, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.053 INFO cells - processing, type: FTCP, cell name: $extract$\FTCP$220, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.054 INFO cells - processing, type: FTCP, cell name: $extract$\FTCP$221, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.054 INFO cells - processing, type: FTCP, cell name: $extract$\FTCP$222, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.054 INFO cells - processing, type: FTCP, cell name: $extract$\FTCP$223, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.055 INFO cells - processing, type: FTCP, cell name: $extract$\FTCP$224, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.055 INFO cells - processing, type: BUFG, cell name: bufg0, top module: top, pass: yosys -> intermediate
Feb 17 03:36:38.055 INFO intermed2input - adding P-term, inputgraph: 0, intermed: 6, name: $auto$coolrunner2_sop.cc:143:execute$253, pass: intermediate -> input
Feb 17 03:36:38.056 INFO intermed2input - adding XOR bits, inputgraph: 6, intermed: 14, name: $auto$coolrunner2_sop.cc:154:execute$254, pass: intermediate -> input
Feb 17 03:36:38.056 INFO intermed2input - adding input bits, inputgraph: 9, intermed: 18, name: $auto$iopadmap.cc:403:execute$263, pass: intermediate -> input
Feb 17 03:36:38.056 INFO intermed2input - adding BUFG, inputgraph: 0, intermed: 29, name: bufg0, pass: intermediate -> input
Feb 17 03:36:38.056 INFO intermed2input - adding register bits, inputgraph: 6, intermed: 20, name: $auto$simplemap.cc:420:simplemap_dff$51, pass: intermediate -> input
Feb 17 03:36:38.057 INFO intermed2input - adding P-term, inputgraph: 1, intermed: 16, name: $auto$coolrunner2_sop.cc:285:execute$260, pass: intermediate -> input
Feb 17 03:36:38.057 INFO intermed2input - adding XOR bits, inputgraph: 8, intermed: 17, name: $auto$coolrunner2_sop.cc:292:execute$261, pass: intermediate -> input
Feb 17 03:36:38.057 INFO intermed2input - adding register bits, inputgraph: 8, intermed: 21, name: $extract$\FTCP$217, pass: intermediate -> input
Feb 17 03:36:38.057 INFO intermed2input - adding P-term, inputgraph: 2, intermed: 5, name: $auto$coolrunner2_sop.cc:143:execute$250, pass: intermediate -> input
Feb 17 03:36:38.058 INFO intermed2input - adding XOR bits, inputgraph: 5, intermed: 13, name: $auto$coolrunner2_sop.cc:154:execute$251, pass: intermediate -> input
Feb 17 03:36:38.058 INFO intermed2input - adding register bits, inputgraph: 5, intermed: 22, name: $extract$\FTCP$218, pass: intermediate -> input
[... 48 lines cut ...]
  17: xc2par::main
             at src/bin/xc2par.rs:198
  18: std::rt::lang_start::{{closure}}
             at /rustc/5e1a799842ba6ed4a57e91f7ab9435947482f7d8/src/libstd/rt.rs:67
  19: std::rt::lang_start_internal::{{closure}}
             at src/libstd/rt.rs:52
  20: std::panicking::try::do_call
             at src/libstd/panicking.rs:292
  21: __rust_maybe_catch_panic
             at src/libpanic_unwind/lib.rs:78
  22: std::panicking::try
             at src/libstd/panicking.rs:270
  23: std::panic::catch_unwind
             at src/libstd/panic.rs:394
  24: std::rt::lang_start_internal
             at src/libstd/rt.rs:51
  25: std::rt::lang_start
             at /rustc/5e1a799842ba6ed4a57e91f7ab9435947482f7d8/src/libstd/rt.rs:67
  26: main
  27: __libc_start_main
  28: _start
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

(engine.rs:1271 is `let zia = result_i.unwrap();`)

rqou, 2020-03-01:
Yes, XC2C64A support is incomplete due to concerns regarding the source of the ZIA mapping table. I may have a solution for this that doesn't require a SEM/FIB, stay tuned

rqou, 2020-03-10:
XC2C64A support has been added now! Let me know if it works for you, and feel free to report further bugs.

closed 2020-03-10 by rqou