azonenberg/openfpga #138: xc2jed2json outputting nonsense

alexforencich, 2020-04-16:
I have a jed file I want to get a netlist for, if possible.  I managed to get xc2jed2json to read it by setting the device field correctly to match the device (xc2c384-10-ft256) and setting the checksum to all zeros, but the returned json is nonsense - it looks like INPUTS_PER_ANDTERM is 40, but many of the ANDTERM instances have way more than 40 inputs - 62, 124, 186, 248, 310, 372.  In fact, I don't think any ANDTERM instance with at least 1 input has 40 or fewer inputs.  Additionally, every ANDTERM input is the same - many repeats of from_iob_0 when converted to verilog, or 2507 in the json.  Any ideas on what might be going wrong here?

rqou, 2020-04-16:
This is because we currently do not have the ZIA/AIM interconnect data for this part (#100 ). We cannot copy the data file from ISE, so we need alternative ways of obtaining this information. One option we have is to take electron micrographs of the required parts of the silicon (http://2.bp.blogspot.com/-DTMX_fGGBfg/UznHhaJ_3nI/AAAAAAAAAjs/pX9eANz3pEM/s1600/zia_vias_cropped.png), and another option is to obtain it by cleverly supplying inputs to the device (https://github.com/rqou/xc2-zia-probe-thingy). Support for the XC2C384 is on the roadmap, but it's not likely to be done soon since none of us have any hardware featuring that chip on hand. If you want to reverse engineer this design before we add official support, you may be able to get the necessary data from ISE yourself.

alexforencich, 2020-04-16:
What JTAG hardware is required for the probing?  I can also send you a board with an XC2C384, if it would help expedite things.

alexforencich, 2020-04-16:
(for context: I am trying to re-use an undocumented board with a Virtex 6 240T FPGA on it that's configured via an XC2C384 CPLD...I have a whole stack of these boards, and would be happy to donate one if it would help get some insight into what the CPLD is doing)

rqou, 2020-04-16:
The existing code (https://github.com/rqou/xc2-zia-probe-thingy/blob/master/poke_device.py) uses a very incomplete "JTAG adapter" firmware (https://github.com/rqou/crabbytty) that is effectively just a bitbang interface. Any other JTAG adapter should be usable as long as it's scriptable from Python and can send INTEST and the custom CPLD programming commands.

A hardware donation would probably help this get done faster, but please do keep in mind that many of us are preoccupied by the current global situation.