azonenberg/openfpga #141: cheat code support for the pocket cores

ThetaX55, 2024-05-02:
Cheats in pocket cores.
In the menu or something.
More savestat support for the cores.

closed 2024-05-02 by azonenberg

ThetaX55, 2024-05-02:
also join to ip for second player.

cyrozap, 2024-05-02:
Assuming they're not a bot, judging by [their repos](https://github.com/ThetaX55?tab=repositories) and their requests (it sounds like they're talking about FPGA-based video game console emulation) I think they're looking for https://github.com/open-fpga.

@ThetaX55, this is a repository for an open source FPGA _toolchain_, not for FPGA-based emulator cores. In fact this repository has absolutely nothing to do with video game emulation, and continuing to comment on this issue is not going to solve whatever issues you're having.